tsea83-g34/TSEA83_Hardware #10: Implement and test ControlUnit component

angloth, 2019-03-29:


angloth, 2019-04-17:
@angloth Implement and test control logic for ALU, RegisterFile and DataForwarding.

@Henkecool Implement and test control logic for DataMemory, (VGA memory).

@jullinator Implement and test control logic for ProgramMemory and WriteBackLogic.

angloth, 2019-05-06:
Control Signals for RegisterFile and DataForwarding are now finished.

angloth, 2019-05-06:
WriteBackLogic and DataMemory control signals are now implemented.

angloth, 2019-05-06:
Todo:
- Implement DataMemory control signals
- Finish ALU control signals
- Double check Programmemory control signals
- Write some testbench for ControlUnit

angloth, 2019-05-06:
Implemented control signals for video memory and programmemory.

jullinator, 2019-05-06:
> Todo:
> 
> * Implement VideoMemory control signals
> * Finish ALU control signals
> * Double check Programmemory control signals
> * Write some testbench for ControlUnit

I can start with the testbench for ControlUnit

angloth, 2019-05-07:
Ok good, I'm working on finishing ALU (#18) and control signals for ALU.

angloth, 2019-05-09:
Control signals for all components have been implemented. Now just need to test them which @jullinator will do for now

angloth, 2019-05-12:
Completed  for now. Will reopen again later if we have bugs,

closed 2019-05-12 by angloth